ARM-software/ebbr #1: Need policy on sharing devices between FW and OS

glikely, 2018-05-15:
EBBR needs to comment on whether it is okay for firmware to access devices at runtime (ie. via a Runtime Services call, or via SecureWorld).

Typically there should be no sharing of devices at all. A device's register interface should be either entirely owned by firmware or entirely owned by the OS.

wmamills, 2018-05-17:
This one can be assigned to me

glikely, 2018-05-17:
@wmamills You should have received an invitation to collaborate from GitHub. You'll need to accept that before I can assign the issue.

wmamills, 2018-05-18:
@glikely I have now.  You assigned the ones I wrote to me so I thought I was grandfathered in somehow.  These permissions are a bit subtle.

glikely, 2018-05-18:
@wmamills Yup, it's a bit weird. I can assign any issue to the person who raised it, but can only assign others if they are part of the project.

commit 14c4600c (2018-07-06): Add requirements on runtime access to devices

To make it absolutely clear that runtime services cannot retain access
to devices that are 'owned' by the OS, add a section spelling the
situation out. Devices may be owned by the OS, or owned by runtime
services, but not both.

Also rework the RTC clock section to allow for the RTC being
inaccessible for exactly that reason.

Resolves: #1  "Need policy on sharing devices between FW and OS"
Resolves: #14  "RTC should be optional"

Reviewed-by: Daniel Thompson <daniel.thompson@linaro.org>
closed 2018-07-06 by glikely