ARM-software/ebbr #14: RTC should be optional

wmamills, 2018-05-17:
Not all embedded platforms have an RTC,  UEFI requires it.
Should it be optional in EBBR?  At least for early levels?
Should we recommend OS actions to deal with this?

daniel-thompson, 2018-05-17:
> Not all embedded platforms have an RTC,  UEFI requires it.
> Should it be optional in EBBR?

Yes, I think this should be optional.

> At least for early levels?

Personally I'd argue this should always be optional.

Including an RTC hits the BoM and implies the addition of a battery or a
largish slow leakage capacitor either of which could easily become the
shortest lifespan component on the board. I think omitting the RTC
is a defensible design decision for some applications and thus IMHO is
not a mistake that EBBR should seek to correct in later levels.

> Should we recommend OS actions to deal with this?

Not sure if this is needed. We'd probably end up describing what Linux
kernel already does w.r.t. time keeping. If we did end up doing
something like that then we would arguably be solving an already solved
problem whilst at the same time elevating Linux's position too high
within an OS agnostic standard.

Daniel.

Uditkr, 2018-05-17:
>That said, even when there is an RTC, it may not be possible to access it via a runtime service. If the RTC is on an i2c bus shared with other devices, then the OS will control the bus and firmware must not try to perform its own transactions.

i2c driver can sit in EL3 and OS or UEFI may do smc call to get i2c  work done.  So exclusive access is maintained.

Uditkr, 2018-05-17:
@daniel-thompson , This is okay to make RTC optional at all level, 
But how we are making doing it i.e. 
1) allowing set/get functions to return an error or 
2) just passing to OS as no-run time 

IMO, with this first Linux will register rtc0 as efi-rtc and rtc1 could be cmos or other sort of RTC. 
is this okay for OS ?

daniel-thompson, 2018-05-18:
> IMO, with this first Linux will register rtc0 as efi-rtc and rtc1 could be cmos or
> other sort of RTC. is this okay for OS ?

I'd prefer to define optional differently.

IMHO we should not require boards to include an RTC (at any level) but that boards with an RTC should expose it via EFI although I don't have particularly strong opinions about whether the "if you have it, expose it via EFI" should kick in at level 0 or level 1.

From Linux point-of-view I don't think it matters. efi-rtc will not probe if the GetTime() returns an error so rtc0 will never be efi-rtc if EFI has no RTC.

commit 14c4600c (2018-07-06): Add requirements on runtime access to devices

To make it absolutely clear that runtime services cannot retain access
to devices that are 'owned' by the OS, add a section spelling the
situation out. Devices may be owned by the OS, or owned by runtime
services, but not both.

Also rework the RTC clock section to allow for the RTC being
inaccessible for exactly that reason.

Resolves: #1  "Need policy on sharing devices between FW and OS"
Resolves: #14  "RTC should be optional"

Reviewed-by: Daniel Thompson <daniel.thompson@linaro.org>
commit aca19fef (2018-10-10): GetTime()/SetTime() should return EFI_UNSUPPORTED when unsupported

UEFI spec is adding ability for all runtime services to return
EFI_UNSUPPORTED if not implemented. Use this return code instead of
EFI_DEVICE_ERROR.

Resolves: #14

Reviewed-by: Peter Robinson <pbrobinson@gmail.com>
closed 2018-07-06 by glikely