ARM-software/ebbr #16: README should help readers not just writters/contributors

wmamills, 2018-05-18:
The current README helps people get started on building the document from source and helping to author it.  This is good and helpful.

However, right up front we should provide guidance for people that find this site that only want to read the current version (or current released version).  Is this just a link to the single *.md file or will it be more complex in the future?  When we have a PDF CI loop we could publish the PDF results also and (under a constant URL)  add a link in the README.

glikely, 2018-05-18:
Completely agree

glikely, 2018-05-24:
Partially resolved, but will leave this issue open until the CI loop is set up and the README can point at the pdf version.

glikely, 2019-04-23:
Closing this issue. CI loop is currently broken, but I'll raise a specific issue just for that. Otherwisre, the README does point to the current document.

closed 2019-04-23 by glikely