ARM-software/ebbr #27: Clarify role of optional UEFI protocols

Googulator, 2018-07-25:
Currently, the specification lists certain UEFI protocols as required, and others optional. Some protocols, for example EFI_GRAPHICS_OUTPUT_PROTOCOL, are listed in neither category. This can be construed such that supporting EFI_GRAPHICS_OUTPUT_PROTOCOL is forbidden.

It needs to be clarified what it means for a protocol to be optional vs. not listed at all.

glikely, 2018-08-09:
Good comment. Thanks. The whole list of UEFI protocols needs to be revamped before a v1.0 release. The current form is a holdover from the SBBR document before UEFI 2.7 was released. The list has not been reviewed in the context of EBBR.

The UEFI spec has its own list of what protocols are required/optional. EBBR aims to be UEFI compliant, but there will be some places of variance due to hardware or software limitations that need to be documented in the spec.

glikely, 2019-03-18:
The entire appendix has been removed as it doesn't add anything to the specification. The UEFI spec already specifies the requirements on an implementation.

closed 2019-03-18 by glikely