ARM-software/ebbr #31: Support for providing a random seed for KASLR

nullr0ute, 2018-09-19:
The UEFI spec allows to provide a random seed to the kernel. Implement the ability to provide a random seed to enable KASLR in the kernel using the UEFI interface. There should be the ability to use a HW entropy source to provide the seed. It might be worthwhile to provide a default software means of generating a random seed but there might be other implications to that.

nullr0ute, 2018-11-21:
An upstream U-Boot commit for some NXP processors retrieves a random seed from the secure monitor mode firmware if it supports it and passes it to the kernel via device tree.

http://git.denx.de/?p=u-boot.git;a=commit;h=a797f274d7ae806d84b9ececf71f043ca6c1502a

glikely, 2018-11-21:
> An upstream U-Boot commit for some NXP processors retrieves a random 
> seed from the secure monitor mode firmware if it supports it and passes 
> it to the kernel via device tree.
> 
> http://git.denx.de/?p=u-boot.git;a=commit;h=a797f274d7ae806d84b9ececf71f043ca6c1502a

That should be easy to wire up to the UEFI random number API (section 
36.5 of the UEFI spec).

g.

nullr0ute, 2019-01-12:
Completely implementation specific, but the sort of functionality implemented here in optee could be used as a source of random entropy on devices without HWRNG https://lwn.net/Articles/775693/

glikely, 2019-04-23:
Should EBBR require EFI_RNG_PROTOCOL to be implemented, or is the upstream UEFI spec language sufficient.

Alternatively, should there be an appendix to EBBR adding additional requirements for secure platforms?

glikely, 2019-04-23:
From monthly meeting - RNG is fundamental enough that EBBR should call it out as recommended -- potentially required for "higher level" compliance in the future

glikely, 2020-08-31:
Discussed at Aug 31st 2020 meeting.

Depends on meta question: do we require add security requirements to EBBR? See notes from:

https://github.com/ARM-software/ebbr/wiki/EBBR-Notes-2020.08.31

glikely, 2020-10-26:
Update from Ilias: Sughosh added RNG protocol, so this should be working now.

- Question: Does this mean RNG should be required in EBBR?
-  Will have effect of kicking a bunch of boards out
- Answer: Must be required for the 'secure' profile of EBBR
  - Strongly recommended on all EBBR platforms if platform has an entropy source

Closing this issue as the infrastructure is in upstream.

closed 2020-10-26 by glikely