ARM-software/ebbr #39: ACPI/DT-agnosticism violation in section 4.2

Googulator, 2019-02-23:
In Section 4.2, we currently have the following recommendation:

> Vendors are recommended to use their Devicetree vendor prefix as their vendor subdirectory name.

Vendors that only produce platforms based on ACPI may not have such a prefix. Since ACPI is the preferred system description method (and the only one supported in SBBR), I suggest adding the words "ACPI vendor ID or" before "Devicetree vendor prefix", to read:

> Vendors are recommended to use their ACPI vendor ID or Devicetree vendor prefix as their vendor subdirectory name.

agraf, 2019-03-02:
The "Devicetree vendor prefix" is identical to U-Boot's CONFIG_SYS_VENDOR. Could you please give an example where someone wants to be EBBR compliant, boot firmware from a shared medium device, have that firmware not be U-Boot and also does not do DT?

glikely, 2019-03-14:
The suggestion is reasonable. EBBR applies equally to U-Boot and Tianocore implementations, show it should apply for platforms that can run Windows

agraf, 2019-03-15:
I agree, but I would like to keep the potential combinations liw. This section talks about the boot rom, so let's ensure we trim the possibilities for divergence down to a minimum.

In general, I don't mind the acpi vendor id. But ket's ensure it's very obvious which one of the two gets picked by the vendor.

glikely, 2019-03-18:
I've added the text, but put it the other way around ("Devicetree vendor prefix or ACPI vendor ID") to acknowledge that the common case is Devicetree. I'm not concerned much either way as this section is only a recommendation and nothing is actually required.

closed 2019-03-18 by glikely