ARM-software/ebbr #47: Create appendix or addendum for secure platform requirements

glikely, 2020-05-21:
From mailing list discussion (courtesy Sumit Garg):

> What do you think of a security addendum or checklist that goes
> alongside EBBR to detail what is required to make the platform actually
> secure?

I would be in favour of such a security addendum.

IMO, "make the platform actually secure" has a bit wider scope than
just secure boot. And the scope may vary depending on the threat model
which may be specific to a particular use-case. So I will try to list
down security features along with their requirements as follows.
Please feel free to extend this list in case I missed any relevant
security feature:

Feature: Secure boot
Platform requirements:
- BootROM being the Root of Trust for Secure boot shall initiate the
chain of trust via verifying the first stage boot-loader.
- Provide platform binding of the root public key used for verification.

Feature: Anti-rollback protection
Platform requirements:
- BootROM shall provide anti-rollback protection for first stage
boot-loader updates.

Feature: Unbrickable firmware updates
Platform requirements:
- BootROM shall support A/B partition scheme to load first stage
boot-loader from alternate locations.

Feature: Secure storage
Platform requirements:
- Either provide a dedicated non-volatile memory accessible to secure
world only or provide a RPMB based secure storage.
- Provide a Hardware Unique Key (HUK) which is accessible to the
secure world only.

Feature: Secure entropy source
Platform requirements:
- Provide a hardware entropy source which is accessible to secure world only.

Feature: Memory firewalls / TZASC
Platform requirements:
- Provide a way to carve out DRAM so that it's accessible to the
secure world only.

Feature: Secure peripherals / TZPC
Platform requirements:
- Provide a way to partition peripherals so that secure peripherals
are accessible to secure world only.

glikely, 2020-09-28:
Discussed in 18 Sep 2020 meeting. Grant to draft a patch that shows how secure requirements will be broken out into a separate profile

xypron, 2020-10-26:
As discussed in 2020-10-26 meeting EFI_RNG_PROTOCOL should be a part of the secure profile.

hrw, 2020-11-14:
SBBR 1.2 has "Secure and Trusted boot" section which is now moved to BBSR (Base Boot Security Requirements) specification.

https://developer.arm.com/documentation/den0107/latest

Maybe that would be handy?

glikely, 2021-02-15:
Closing this issue as BBSR will probably be the way these requirements come in instead of adding them directly to EBBR

closed 2021-02-15 by glikely