ARM-software/ebbr #56: RISC-V: privilege level when entering OS

xypron, 2020-09-21:
The UEFI spec has this sentence:

"When UEFI firmware handoff control to OS, the RISC-V is operated in machine-mode privilege." (M-mode is the equivalent to EL3 in ARM). This does not make any sense to me when using a secure execution environement (SEE) like OpenSBI.

The hand-off should occur in S-Mode if the CPU supports it and only in M-Mode when the CPU only supports M-mode.

We should prescribe this in the EBBR and somehow get the UEFI spec fixed afterwards.

glikely, 2020-09-23:
Can this be raised at the UEFI spec now? I cannot see any barriers to getting this ECR raised and accepted. Then we can talk about wether or not to carry similar language in EBBR for the short term

xypron, 2020-09-23:
Abner Chang wrote on the boot-architecture list that he raised an ECR requiring S-mode but that it is not yet approved. We could use the sentence from his mail.

glikely, 2021-02-15:
Closing for lack of progress. Can be reopened if work is resumed

closed 2021-02-15 by glikely