ARM-software/ebbr #57: RISC-V: calling convention

xypron, 2020-09-21:
UEFI 2.8B spec chapter "2.3.7.3 Detailed Calling Convention" does not describe which registers are saved by the UEFI payload's entry point and restored by the payload before calling the UEFI API or returning to the UEFI payload. This concerns especially
registers gp (x3) and tp (x4).

Into the EBBR or UEFI spec we should put a link to the "RISC-V ELF psABI specification" https://github.com/riscv/riscv-elf-psabi-doc/blob/master/riscv-elf.md which is referenced by "The RISC-V Instruction Set Manual".

From the "RISC-V ELF psABI specification" one might conclude that the UEFI payload should not be allowed to change gp and tp before calling ExitBootServices() or SetVirtualAddressMap() whichever occurs last.

Due to this missing clarification U-Boot is currently saving gp before calling the entry point of the payload and restores it on reentry or on entry of an API call. Nothing is done for tp.

glikely, 2020-11-23:
Action: Grant to look through the patch again with a view for merging

glikely, 2021-02-15:
Closing for lack of progress. Can be reopened if work is resumed

closed 2021-02-15 by glikely