ARM-software/ebbr #61: Exclude EFI_DECOMPRESS_PROTOCOL

xypron, 2020-10-24:
The UEFI specifiction does not describe the compression algorithm used for the EFI_DECOMPRESS_PROTOCOL. Only sample code is provided. So it is impossible to create an independent implementation.

BaseTools/Source/C/Common/EfiCompress.c has a BSD-2-Clause-Patent license. It is unknown if an independent implementation of the compression protocol would infringe Intel's patents.

We should exclude the requirement to implement the EFI_DECOMPRESS_PROTOCOL from the EBBR specification.

daniel-thompson, 2020-10-26:
> The UEFI specifiction does not describe the compression algorithm used
> for the EFI_DECOMPRESS_PROTOCOL. Only sample code is provided.

Can you explain what you mean by only sample code is provided? There is sample code in the appendix but I count around 12 pages describing the compression/decompression algorithm and these have been included in the spec since at least 2006.

glikely, 2020-10-26:
Agree, this protocol can probably be dropped from EBBR

commit 72f3e2d1 (2021-02-15): Override UEFI section 2.6 requirements

EBBR only requires a subset of UEFI. Provide a replacement for the UEFI
section that lists base requirements.

Fixes: #60
Fixes: #61
Fixes: #64

Reviewed-by: Heinrich Schuchardt <xypron.glpk@gmx.de>
closed 2021-02-15 by glikely