ARM-software/ebbr #63: Variable update at runtime

xypron, 2020-11-06:
In chapter 4 this requirement is made:

"Firmware needs a method to modify variable storage at runtime while the OS controls access to the device."

In the footnote: "If firmware doesn't have a dedicated channel to the storage device, then the OS must proxy all runtime storage IO."

- The requirement is superfluous if we do not require SetVariable() to be implemented.
- It is completely unclear how an OS could proxy runtime storage IO.

The requirement should be removed.

glikely, 2020-11-16:
There is a working implementation of runtime SetVariable() that passes the requests to StMM running under OPTEE, that then uses the normal world optee-supplicant to commit the changes to flash.

It is very important for distros. I'm still on the fence on whether or not it should be required. Not all platforms will be able to implement it, but there really isn't a great alternative at this time without re-architecting distro installers.

glikely, 2021-03-15:
Comment from today's EBBR biweekly (Loic Minier): Are there minimum size requirements on variable store?

There are no requirements at this time. Something to investigate. Bill remembers seeing 64k somewhere. Should open another issue to track this

xypron, 2021-03-15:
Cf. #77

closed 2023-08-29 by xypron