ARM-software/ebbr #64: Globally Defined Variables

xypron, 2020-11-06:
Chapter "3.3 Globally Defined Variable" of the UEFI spec defines possible global variables.

The chapter is not referenced in the EBBR.

We need a definition which variables must exist on an EBBR compliant device and which must be supported.

glikely, 2020-11-16:
I imagine EBBR wouldn't change the definition of global variables, correct? A section could be added for "Required Global Variables". What variables do you see as necessary?

the obvious:
- Boot####
- BootCurrent
- BootNext
- BootOrder
- BootOptionSupport

glikely, 2020-11-16:
Hmmm... maybe not that last one; BootOptionSupport would be trivial to implement, but how is it actually used?

commit 72f3e2d1 (2021-02-15): Override UEFI section 2.6 requirements

EBBR only requires a subset of UEFI. Provide a replacement for the UEFI
section that lists base requirements.

Fixes: #60
Fixes: #61
Fixes: #64

Reviewed-by: Heinrich Schuchardt <xypron.glpk@gmx.de>
closed 2021-02-15 by glikely