ARM-software/ebbr #68: EFI_DT_FIXUP_PROTOCOL

xypron, 2020-12-14:
Do we want to include the EFI_DT_FIXUP_PROTOCOL in the EBBR (https://github.com/U-Boot-EFI/EFI_DT_FIXUP_PROTOCOL)?

glikely, 2021-01-16:
Depends on how you want to specify it in EBBR. I'm not keen on adding it to the requirements until we know the relevant projects (Yocto, OpenWRT, Buildroot, etc) will adopt it.

What are you thinking for specification text?

xypron, 2021-01-16:
Am 16. Januar 2021 10:31:45 MEZ schrieb Grant Likely <notifications@github.com>:
>Depends on how you want to specify it in EBBR. I'm not keen on adding
>it to the requirements until we know the relevant projects (Yocto,
>OpenWRT, Buildroot, etc) will adopt it.

Hello Grant,

They will be using the U-Boot source where I have enabled the protocol.

To me the relevant project is GRUB. I have analyzed what code needs changing there but not implemented anything yet.

(OpenWRT does not use GRUB.)

I had hoped for feedback in the DTE project where the idea emerged. But in the last meeting I was referred to EBBR.

>
>What are you thinking for specification text?

I would copy the readme from https://github.com/U-Boot-EFI/EFI_DT_FIXUP_PROTOCOL.

Best regards

Heinrich

jan-kiszka, 2022-04-28:
What's the status of this? U-Boot supports it now, systemd picked it up, EFI Boot Guard will as well - but we may also need it in alternative UEFI providers (e.g. edk2).

jan-kiszka, 2022-06-24:
Second ping. We are relying in this interface by now to enable secure boot via a unified kernel image for Debian (and conceptually also other distros) while permitting replacement DTBs in that unified kernel image.

https://github.com/siemens/efibootguard/blob/master/docs/UNIFIED-KERNEL.md
https://github.com/siemens/efibootguard/blob/master/kernel-stub/fdt.c
https://gitlab.com/cip-project/cip-core/isar-cip-core

xypron, 2022-06-24:
@glikely Could you, please, send out an invite for the next EBBR meeting to discuss this protocol?

jan-kiszka, 2022-09-07:
Did this discussion take place?

jan-kiszka, 2022-09-19:
@xypron, I heard from @glikely that someone else is driving EBBR now - but I forgot the name :frowning_face:.

daniel-thompson, 2022-09-20:
It is on the agenda for the upcoming meeting: https://lists.linaro.org/archives/list/boot-architecture@lists.linaro.org/thread/N6BPBGHT5U2LOZRBSMIVIVLFTIISBXFR/

jan-kiszka, 2022-09-29:
I've found another use case where this protocol would be very beneficial - rather than the fragile approach that is currently implemented in a BSP. See https://github.com/U-Boot-EFI/EFI_DT_FIXUP_PROTOCOL/issues/1.

xypron, 2022-10-11:
Am 19. September 2022 14:27:23 MESZ schrieb Jan Kiszka ***@***.***>:
***@***.***, I heard from @glikely that someone else is driving EBBR now - but I forgot the name :frowning_face:.
>

Vincent Stehlé ***@***.***> organises the EBBR meetings now. There will be a meeting next Monday.

Best regards 

Heinrich

jan-kiszka, 2023-11-10:
More than a year has passed - any progress with this?

vstehle, 2025-06-18:
Call of Jun 18: let's close for now.

jan-kiszka, 2025-06-19:
https://github.com/ARM-software/ebbr/wiki/EBBR-Notes-2025.06.18 - who will drive this for UEFI then?