ARM-software/ebbr #72: Uniqueness of device paths

xypron, 2021-02-03:
Device paths are used to identify devices and therefore should be unique.

The UEFI spec requires that InstallMultipleProtocolInterfaces() checks that it does not installing a duplicate device paths. It has to return an EFI_ALREADY_STARTED in this case.

For InstallProtocolInterface() such a requirement is missing. So we could end up creating conflicts.

The UEFI spec should be amended.

glikely, 2021-02-03:
Sounds like something that should be fed back to the USWG instead of put
into EBBR.

g.

glikely, 2021-02-12:
Is this something we need to discuss at the biweekly on Monday?

xypron, 2021-02-12:
On 2/12/21 8:01 PM, Grant Likely wrote:
> Is this something we need to discuss at the biweekly on Monday?
>

We should discuss how we can carry such observations to the right
UEFI.org bodies.

glikely, 2021-02-15:
Closed as it is a UEFI USWG topic

closed 2021-02-15 by glikely