ARM-software/ebbr #84: Firmware Storage: Partition type GUID recommendations

samueldr, 2021-05-09:
Hi,

From the *Firmware Storage* chapter:

https://github.com/ARM-software/ebbr/blob/6dae6fd764e7a0a1111943c41cc3eb7eef3012b0/source/chapter4-firmware-media.rst#L114-L116

I couldn't find a partition type matching something like a *firmware protective partition* in the few lists I know of:

 - https://en.wikipedia.org/wiki/GUID_Partition_Table#Partition_type_GUIDs
 - https://sourceforge.net/p/gptfdisk/code/ci/79977a7e48ad6da9f206005abe3c2553402d8076/tree/parttypes.cc
 - [UEFI spec](https://uefi.org/sites/default/files/resources/UEFI_Spec_2_8_final.pdf) chapter 23 (three entries) 

Is there a non-binding recommendation for such a partition type? Did I miss a source of well-known partition types?

Should the spec recommend (and own) a specific partition type GUID?

Thank you!

samueldr, 2022-01-24:
As far as Tow-Boot is concerned, which sparked this question, the assumption has been that the project itself owns the Type GUID.

As such, the partition type `"67401509-72E7-4628-B1AF-EDD128E4316A"` represents the protective partition used by the Tow-Boot project when installing a platform firmware to a shared storage area.

 - https://github.com/Tow-Boot/Tow-Boot/blob/eb2a83466e220e31b561955b2c59b9fc9c21cd46/support/overlay/overlay.nix#L112-L114

daniel-thompson, 2022-01-26:
Assuming this GUID ID is one you have generated yourself then it will meet the spec.

The theory is that the RequiredPartition flag is the key piece of information to help automatic partitioning tools make good decisions (e.g. don't delete the partition). The requirement to use "a" use a GUID that identifies it as a firmware protective partition is to prevent existing GUIDs being used since they might undermine the confidence in the RequiredPartition flag (e.g. if the firmware happened to use a FAT filesystem internally for variable storage then this is fine... but don't give it a GUID that makes an OS try to mount it).

Thus any party can create a GUID and providing they define what it means (it contains some firmware that is necessary to boot) then it will be "a GUID that identifies it as a firmware protective partition".

PS Personally I'd like there to be a standard or de-facto standard fallback value so that firmware that has no opinions on what GUID to use (e.g. no reason to pick a different one) can pick a common one. This is mostly because having a well known value gives us something to add to partitioning tools so they can present a user-friendly name when they see it.

trini, 2022-01-26:
To what granularity should there be GUIDs here?  For example, should U-Boot as an upstream project generate one GUID and use it for all platforms?  Or one GUID per SoC / SoC family where the firmware magic locations are the same?  And then what would be the guidance for customized distributions of U-Boot such as Tow-Boot ?

daniel-thompson, 2022-01-26:
> To what granularity should there be GUIDs here?

Currently I don't think the spec offers any guidance on this.

Naturally I have an opinion but not an authoritative one! FWIW it is mostly in my PS: few GUIDs are better than many... and one is better than few.

samueldr, 2022-01-26:
It would be convenient if there was a default "EBBR Compliant Firmware" GUID type. I figure it would be easier for partitioning utilities to know about a single GUID, than even a handful of GUIDs (per project).

The spec could be worded in a way that makes it the fallback type when another one is not *required* by the platform. Similarly to how `0xF8` is used with MBR, except if "an immutable feature of the platform makes this impossible". Or worded in a way that makes it an option?

I don't mind either way. I see now that the `RequiredPartition` is what matters.

xypron, 2022-11-18:
Firmware may require multiple partitions with distinct GUIDs.

U-boot may use partition type b000 "U-Boot boot loader "3DE21764-95BD-54BD-A5C3-4ABE786F38A8" for storing the environment. Some devices require U-Boot SPL living in one partition and OpenSBI, device-tree, and main U-Boot (as a FIT image) living in the other partition. Secure world may add a few more partitions.

EBBR may make recommendations for the GUIDs but I think a system that uses vendor GUIDs for its firmware partitioning scheme should not be blemished as non-compliant.

closed 2023-01-16 by vstehle

xypron, 2023-01-16:
The flag is described in the UEFI specification, but I don't see how it is set via gdisk.