ARM-software/ebbr #143: Make SetVariableRT mandatory

apalos, 2025-06-24:
SetvariableRT was never mandatory and it has been causing a lot of problems wrt to distro installers and capsule update on-disk. 

https://github.com/rhboot/efivar/pull/282 is the last piece of the puzzle for u-boot to support setvariable RT regardless of the storage. I think we should in the future make it mandatory

xypron, 2025-06-24:
Hello Ilias,

I think this is a valid discussion to have. I confirm that for a distro life would be easier with SetVariable at runtime.

There is no UEFI service called SetVariableRT(). We should refer to SetVariable() at runtime.

If we opt to require SetVariable(), we should also require GetVariable() and GetNextVariable() and possibly QueryVariableInfo() at runtime

With https://github.com/rhboot/efivar/pull/282 there are now two hacked up ways in Linux to implement SetVariable() which both will not work between ExitBootServices() and the operating system being fully set up. Are other relevant operating systems, e.g. BSD and Windows on ARM, expected to support these hacks?

Two weeks ago we discussed if we should allow systems without any  runtime time services, specifically without virtual memory services, to be called EBBR compliant  (#141) out of security considerations. You now walk into the opposite direction proposing a requirement for run time services after SetVirtualAddressMap(). Have the previous security concerns been misled?

Could you, please, elaborate why you don't like the current capsule on disk implementation in U-Boot (which is ignoring flag EFI_OS_INDICATIONS_FILE_CAPSULE_DELIVERY_SUPPORTED  in OsIndications).

Best regards

Heinrich

fozog, 2025-06-25:
HelloI believe the key question for RT mandatory function is: what is the minimal set of functions required to deal with firmware upgrade commit and rollback ?CheersFF

With rhboot/efivar#282 there are now two hacked up ways in Linux to implement SetVariable() which both will not work between ExitBootServices() and the operating system being fully set up. Are other relevant operating systems, e.g. BSD and Windows on ARM, expected to support these hacks?

apalos, 2025-06-25:
> HelloI believe the key question for RT mandatory function is: what is the minimal set of functions required to deal with firmware upgrade commit and rollback ?CheersFFOn 25 Jun 2025, at 01:48, 

Hi Francois, if you are referring to https://github.com/apalos/edge-iot-arch-guide/blob/main/source/FWU/MBFW/chapter1-introduction.md then SetVariable at runtime isn't needed. The commit/rollback functionality is triggered by an empty capsule. However, we did have discussions on controlling it via a variable if it's supported